mit-acl/faster #2: FASTER with Dynamic Obstacles

singhh23, 2020-05-08:
Hi, this is not an issue per-se, but I was wondering if there's an approach which could be used to introduce dynamic obstacle handling into this framework.

Apologies if there is a better forum for this.

jtorde, 2020-05-12:
Hi H Singh,

So FASTER is designed mainly for static worlds. There are several ways that you could try to make it work in a dynamic environment:

- If the robot is moving much faster than the obstacles, FASTER would probably work (since from the robot's perspective, the obstacles are barely moving). However, no safety guarantees can be made at all times. In real-world experiments you may have also problems with blur in the depth image of the camera. 

- In cases where the velocities of the robot and the obstacles are similar, here are some thoughts: 

       - You may need a mapper that handles dynamic obstacles, so that these obstacles are not "replicated" as they move along. 

        - Another question is how you model the obstacles. Do you know their trajectory? Do you know their shape? How are you going to model the obstacle avoidance constraint? How are these trajectories encoded in your global and local planner? For the global planner one option would be to add one dimension (x,y,z,**t**) to the search space for example. 

Hope this helps!  Closing it for now, but let me know if you have more questions!

closed 2020-05-12 by jtorde

jtorde, 2021-08-04:
Just un update on this, in case it helps: You may be interested in checking out MADER or PANTHER (trajectory planners with dynamic obstacles):

[https://github.com/mit-acl/mader](https://github.com/mit-acl/mader)
[https://github.com/mit-acl/panther](https://github.com/mit-acl/panther)

singhh23, 2021-08-09:
Thanks @jtorde, MADER seems to be towards what I'm looking for - very interesting work!