mit-acl/faster #3: ROS Melodic compatibility changes

betaBison, 2020-05-18:
I'm trying to get Faster working on my machine with Ubuntu 18.04 and ROS Melodic. What I've changed so far:
- I installed the "ros-melodic-" dependencies instead of the "ros-kinetic-" dependencies
- I'm using the "feature/support_ros_melodic" branch of acl-gazebo instead of the "master" branch

With these changes, the workspace builds with the following two warnings:
```
Warnings   << acl_sim:cmake /home/user/faster_ws/logs/acl_sim/build.cmake.000.log
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/OGRE/cmake/modules/FindOGRE.cmake:318 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:175 (find_package)
  CMakeLists.txt:21 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
```
and
```
Warnings   << decomp_ros_utils:cmake /home/user/faster_ws/logs/decomp_ros_utils/build.cmake.000.log
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'decomp_util' but neither
  'decomp_util_INCLUDE_DIRS' nor 'decomp_util_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  /home/user/faster_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:214 (catkin_package)
  CMakeLists.txt:46 (cs_export)
```
For the aerial robot, when I press "START", it takes off, but when I give it a nav goal, nothing happens.
The last thing that prints in the terminal where I ran `roslaunch faster faster.launch` is
```
[ INFO] [1589819914.036287779]: Waiting for world to camera transform...
```
The terminal where I ran ` roslaunch global_mapper_ros global_mapper_node.launch` has this persistent warning:
```
[ WARN] [1589820037.314969058, 135.148000000]: [world_database_master_ros] OnGetTransform failed with "Q01s" passed to lookupTransform argument source_frame does not exist. 
```
In the terminal where I ran `roslaunch acl_sim start_world.launch` has an error:
```
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
```
and a warning:
```
[ WARN] [1589819857.395608542, 66.828000000]: dynamic reconfigure is not enabled for this image topic [camera/rgb/image_raw] becuase <cameraName> is not specified
```

Similarly, when I try running the ground robot, the robot never moves after I send nav goals.  
The last thing that prints in the terminal where I ran `roslaunch faster faster.launch quad:=JA01` is 
```
[ INFO] [1589819442.228136026]: Waiting for world to camera transform...
```

There's this repeated warning in the terminal where I ran `roslaunch global_mapper_ros global_mapper_node.launch quad:=JA01`
```
[ WARN] [1589819328.864921609, 63.361000000]: [world_database_master_ros] OnGetTransform failed with "A01" passed to lookupTransform argument source_frame does not exist. 
```
The terminal where I ran ` roslaunch faster ground_robot.launch` has the same REST request error and dynamic reconfigure warning as the aerial vehicle.

I appreciate any suggestions you have for me!

jtorde, 2020-05-18:
Hi @betaBison!

I've just pushed to the master branch all the fixes needed to support both ROS Melodic and ROS Kinetic, and tested it in two different machines with these two ROS versions.

The error "Error in REST request" stills appears, but you can safely ignore it (see [this](https://answers.gazebosim.org//question/22263/error-in-rest-request-for-accessing-apiignitionorg/))

Let me know if you have any more issues!

betaBison, 2020-05-18:
Wow, impressive response time @jtorde! Both the aerial and ground robots start moving for me now. The only thing I had to change for my setup was to use the HTTPS url instead of ssh in [faster_ground_robot.rosintall](https://github.com/mit-acl/faster/blob/master/faster/install/faster_ground_robot.rosinstall)

I'm excited to start exploring!

closed 2020-05-18 by betaBison