mit-acl/faster #6: RVIZ No fixed frame [world] Problem

almacitunaberk, 2020-06-27:
First of all, I want to thank everyone who contributed to this beautiful project.

I use Ubuntu 18.04 and ros-melodic on my machine. 

I did everything in the right order. However, I am having an RVIZ problem:
It says that there is no Fixed Frame [world] does not exist. 
I would really appreciate if any one could help.

jtorde, 2020-06-27:
Hi @almacitunaberk! So I'm not able to reproduce this error on my machine, and I think this error shouldn't happen if all the commands are run in the right order... Could you take an screenshot of the TF tree? It should look like this:
[image]

This is how the default Rviz visualization should look like:

[image]

Let me know how the TF looks in your case, to see if I can help further

Thanks!

jtorde, 2020-07-04:
Closing this for now, but feel free to open it again if you still have this problem

Thanks!

closed 2020-07-04 by jtorde