mit-acl/faster #7: roslaunch faster faster_interface.launch quad:=JA01 is_ground_robot:=true show "Not initialized yet"

YouShaoze, 2020-06-30:
I follow the steps to configure and type four commands ，and then click the three buttons in the gui that show "Not initialized yet".
Then there is no message generated or sent by goal or cmd_vel. 

Terminal display information :

... logging to /home/ysz/.ros/log/55eb0e4c-bac2-11ea-b734-000c29a4c471/roslaunch-ysz-35817.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ysz:43171/

SUMMARY
========

PARAMETERS
 * /JA01/faster_commands/is_ground_robot: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    rqt_gui (rqt_gui/rqt_gui)
    rviz (rviz/rviz)
  /JA01/
    faster_commands (faster/faster_commands.py)

ROS_MASTER_URI=http://localhost:11311

process[JA01/faster_commands-1]: started with pid [35835]
process[rviz-2]: started with pid [35836]
process[rqt_gui-3]: started with pid [35837]
('self.is_ground_robot=', True)

(when I click the button in the gui)
Not initialized yet
Not initialized yet
Not initialized yet
Not initialized yet

jtorde, 2020-07-09:
Hi @YouShaoze! 

I'm unable to reproduce your errror in my machine. Could you try to repeat the steps specified in the readme? Note that you only have to press the button "Start" on the GUI. 

Let me know if you still have this issue after this. 

Thanks!

YouShaoze, 2020-07-15:
[image]
Here's the interface I run the demo. 
roslaunch faster ground_robot.launch          Code running well
roslaunch global_mapper_ros global_mapper_node.launch quad:=JA01     Code running well
roslaunch faster faster_interface.launch quad:=JA01 is_ground_robot:=true   Not initialized yet
roslaunch faster faster.launch quad:=JA01    Code running well

YouShaoze, 2020-07-15:
> Hi @YouShaoze!
> 
> I'm unable to reproduce your errror in my machine. Could you try to repeat the steps specified in the readme? Note that you only have to press the button "Start" on the GUI.
> 
> Let me know if you still have this issue after this.
> 
> Thanks!

Hi @jtorde 
I found the cause of the failure, 
I didn't install pyquaternion correctly.
my python version is 2.7
So it's not appropriate to install 
pip install pyquaternion
I used
python -m pip install pyquaternion
and succeed.

jtorde, 2020-07-15:
Hi @YouShaoze 

Thanks, I've just modified the readme. 
`pip install pyquaternion` may or may not work depending on the value of `PATH`(see [this](https://snarky.ca/why-you-should-use-python-m-pip/). 

Thanks!

closed 2020-07-15 by jtorde