mit-acl/faster #8: protobuf version conflict

Nianan0505, 2020-07-10:
My system is ubuntu16.04 gazebo8 and protobuf version is 3.6
When using FASTER with an aerial robot
the step **catkin build**  `the` error occurs as follows:

```
 #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /usr/include/ignition/msgs0/ignition/msgs/geometry.pb.h:30:0,
                 from /usr/include/ignition/msgs0/ignition/msgs/collision.pb.h:28,
                 from /usr/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:16,
                 from /usr/include/ignition/msgs0/ignition/msgs.hh:3,
                 from /usr/include/ignition/transport3/ignition/transport/Node.hh:32,
                 from /usr/include/gazebo-8/gazebo/physics/Entity.hh:25,
                 from /usr/include/gazebo-8/gazebo/physics/Model.hh:30,
                 from /usr/include/gazebo-8/gazebo/physics/Actor.hh:24,
                 from /usr/include/gazebo-8/gazebo/physics/physics.hh:2,
                 from /home/yjq/program/uav_ws/src/acl-gazebo/acl_sim/src/disable_physics_kinetic.cpp:4:
/usr/include/ignition/msgs0/ignition/msgs/cylindergeom.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/ignition/msgs0/ignition/msgs/cylindergeom.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/ignition/msgs0/ignition/msgs/cylindergeom.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
```
**Which version should the protobuf is?**
Thank you sooo much for your reply.

jtorde, 2020-07-10:
Hi @Nianan0505!

So running `protoc --version` on my Ubuntu 16.04 machine says the protoc version installed is 2.6.1. 

I think (and reading [this](https://answers.gazebosim.org//question/18096/this-file-was-generated-by-an-older-version-of-protoc/)) Gazebo is built againt the `protoc` version released to the ubuntu repos (which is 2.6.1 for Ubuntu 16.04). 

Could you try to install Gazebo 7.0.0 (the one [recommended for ROS Kinetic](http://gazebosim.org/tutorials/?tut=ros_wrapper_versions)) and protoc version 2.6.1 instead? In the second answer of [this question](https://answers.gazebosim.org//question/18096/this-file-was-generated-by-an-older-version-of-protoc/) there are some steps that may be helpful to do this. 

Thanks

YouShaoze, 2020-07-15:
Hi， @jtorde  @Nianan0505 
I have the same problem, and I use the husky A200 mobile platform. The protoc version drop to 2.6.1 will result in many other software configuration errors and failures, so the best way is not to install gazebo simulation modules. I met this problem and did this on the robot's computer, compiled correctly and passed......

Nianan0505, 2020-07-28:
> Hi @Nianan0505!
>
> So running `protoc --version` on my Ubuntu 16.04 machine says the protoc version installed is 2.6.1.
>
> I think (and reading [...]

Thank you so much for your help.
I installed Gazebo 7, and did not meet this problem.

closed 2020-07-28 by jtorde