mit-acl/faster #9: I used faster on the ground robot, but the replanning process always failed

Nianan0505, 2020-08-12:
Hi, I wanted to use FASTER on the ground robot. And I run the "roslaunches" according to the exact procedure.
When I first clicked the "start" and give the goal by "2D Nave Goal", it run well and reached the destination.
But it failed when I tried to give the next goal by "2D Nave Goal".
The terminal is:

```
No replanning needer because
status_=YAWING
```
And the car is really swaying in the simulation environment.
Can you help me solve this problem? Thanks a lot.
[image]

jtorde, 2020-11-06:
Hi @Nianan0505!

Yes, you are right. I think changing the hand-coded parameters of [this file](https://github.com/mit-acl/faster/blob/master/faster/scripts/goal_odom_to_cmd_vel_state.py) (probably the values in [this line](https://github.com/mit-acl/faster/blob/master/faster/scripts/goal_odom_to_cmd_vel_state.py#L152) or [this one](https://github.com/mit-acl/faster/blob/master/faster/scripts/goal_odom_to_cmd_vel_state.py#L147)) will probably solve this issue. 

Feel free to do a pull request if you are able to solve it

codespring-lh, 2020-11-13:
> Hi, I want to use FASTER on the ground robot. And I run the "roslaunches" according to the exact procedure.
> When I first click the "start" and give the goal by "2D Nave Goal", it runs well and reached the destination.
> But for the next goal, it always fails and the car is swaying in the simulation environment.
> 
> Do you know the possible causes of this problem? Thanks a lot.
> What's more, I'd like to know how the car knows its exact real-time position. Thanks again!

Hi,I met the same problem,have you solved it?

codespring-lh, 2020-12-22:
> Hi @Nianan0505!
>
> Yes, you are right. I think changing the hand-coded parameters of [this file](https://github.com/mit-acl/faster/blob/master/faster/scripts/goal_odom_to_cmd_vel_state.py) (probably [...]

Hi,I have done some experiments using Faster in the real world and it did works but when i set the next goal the car can't move.So,I want to know how to modify these hand-coded parameters of goal_odom_to_cmd_vel_state.py,increase them or decrease them?Hope for your reply.

jamiie1999, 2023-04-21:
> Hi,I have done some experiments using Faster in the real world and it did works but when i set the next goal the car can't move.So,I want to know how to modify these hand-coded parameters of [...]

hello,i am also having some problems,have you solved?

qimao7213, 2024-12-15:
Hi, have you know how to fix these parameters?