mit-acl/faster #10: Intel Realsense D435 is ready and I want to do some expremients on the real ground robot using FASTER.

codespring-lh, 2020-10-21:
Hi, I want to use FASTER on the real ground robot. And I have installed realsense-ros package,and i also have changed some topics in the faster.launch file.But the robot is still stopped.

Do you know the possible causes of this problem? Thanks a lot.
[image]

jtorde, 2020-11-06:
Hi @codespring-lh!

The answer to this issue depends a lot on the ground robot you have. Do you have a Jackal or a different robot? If you have a Jackal, [here](http://wiki.ros.org/Robots/Jackal#Videos:~:text=Videos) is a detailed explanation of how to do the set up. Essentially, you need to run one node(s) on the real robot that subscribe to the topic cmd_vel and that sends the motor commands to the robot.

For the Realsense, I'd make sure that the [mapper](https://gitlab.com/mit-acl/lab/acl-mapping) is receiving correctly to the depth image obtained by the depth camera, [rqt_graph](http://wiki.ros.org/rqt_graph) will probably show this. 

Hope this helps!

codespring-lh, 2020-11-06:
> Hi @codespring-lh!
>
> The answer to this issue depends a lot on the ground robot you have. Do you have a Jackal or a different robot? If you have a Jackal, [...]

Thanks for your reply!I don't have a jackal but i have a replaced one named kobuki.And i did publish the topic cmd_vel through Faster and the kobuki did move.But there is still some questions.First,the robot sometimes moves so fast and the map is too dense.
About the Realsense,I have two devices,the one is D435i and the other is t265(supply odom data,i noticed that goal_odom_to_cmd_vel_state.py need odom data),and the mapper is receving corrctely to the depth image and  can correctly displayed on rviz.
Below are my start setup and some pictures of rviz or launch files.
   1:roslaunch kobuki_node minimal.launch(start the chassis)
   2:roslaunch realsense2_camera rs_d400_and_t265.launch
   3:roslaunch global_mapper_ros global_mapper_node.launch quad:=JA01
   4:roslaunch faster faster_interface.launch quad:=JA01 is_ground_robot:=true
   5:roslaunch faster faster.launch quad:=JA01
Hope for your reply.
[image]

[image]
[image]
[image]

xdwgood, 2020-12-10:
@jtorde  How to transfer the deep data released by d435 to `faster`?
I mean: I have a d435, How do I establish a connection between d435 and `faster`? Are there any reference documents? Thank you

jtorde, 2020-12-10:
Hi @xdwgood  and @codespring-lh!

I've just added a section at the end of the readme explaining how to use the mapper and the planner with a real camera. 

Let me know if you still have any issues!

xdwgood, 2020-12-14:
@jtorde  Thank you for your help. After I changed the lanuch file according to the instructions you added, I got the following:

First run:`roslaunch realsense2_camera rs_camera.launch `

then`roslaunch global_mapper_ros global_mapper_node.launch`

[image]

then `roslaunch faster faster_interface.launch`

[image]

then`roslaunch faster faster.launch`

then I use `rqt_graph`

[image]

My question is, how can I display the real world in rivz (I can move the camera, the blue grid shown in Rviz is the unknown space and the orange one is the occupied- known space))

jtorde, 2021-03-02:
I'm closing this for now, feel free to reopen if you still have this issue

closed 2021-03-02 by jtorde