mit-acl/faster #11: ERROR: cannot launch node of type

Jasonleeo-cpu, 2020-11-05:
ERROR: cannot launch node of type [faster/faster_node]: can't locate node [faster_node] in package [faster],when I run roslaunch faster faster.launch,how can I solve this problem?

jtorde, 2020-11-06:
Hi @Jasonleeo-cpu! 

Some quick questions so that I can help better with this issue:

- Did  you add `source PATH_TO_YOUR_WS/devel/setup.bash`  to your ~/.bashrc, and then run `source ~/.bashrc` in the terminals (or, equivalently, close and open new terminals)?

- What is the output of running `rospack find faster`?

- In your ~/.bashrc, are you doing first `source /opt/ros/melodic/setup.bash` and then 'source PATH_TO_YOUR_WS/ws/devel/setup.bash' or the other way around?

Jasonleeo-cpu, 2020-11-07:
Thanks, I tried it and it helps, I also find a problem that there are some files had wrong names in this code so that it can't create the file name called faster_node, hope it can help others.

closed 2020-12-01 by jtorde