mit-acl/faster #12: How to programmatically set goal point for drone

jcyriac2, 2020-11-22:
Hi, great work on this novel planner! I was wondering whether if there is a way to easily set the goal point for the planner without having to click a point in Rviz? I wanted to perform some repeatable experiments with a constant goal point. Will publishing on the `/SQ01s/goal` topic do the trick or is there a way to do this in the existing code base that I am unaware of?

Thanks!

jtorde, 2020-11-23:
Hi @jcyriac2!

To set the (terminal) goal for the planner, you need to publish to the topic `/move_base_simple/goal`. Here is one example that you can use from the terminal:

`rostopic pub /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: now, frame_id: "map"}, pose: {position: {x: 10.0, y: 0.0, z: 0.0}, orientation: {w: 1.0}}}'`

You can also publish this easily from [C++](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29) or [Python](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29)

Hope this helps!

jcyriac2, 2020-11-23:
Yes, this is perfect! Thank you @jtorde

closed 2020-11-23 by jcyriac2