mit-acl/faster #13: Wrong findGUROBI

akshitj1, 2020-12-02:
https://github.com/mit-acl/faster/blob/3af4e9b276452694b6d527f13eaac926bef0138b/faster/FindGUROBI.cmake#L24
In the findGUROBI, both GUROBI_LIBRARY and GUROBI_CXX_LIBRARY matches to same `libgurobi_c++.a` as same pattern is present in both. This results to not linking of `.so` files eg. `libgurobi91.so`. This leads to undefined reference errors in gurobi.

jtorde, 2020-12-02:
Hi @akshitj1

Could you try to compile it again with the last commit (fc19020)?
Let me know if that still doesn't solve the problem

jtorde, 2020-12-02:
Also, remember to do [this](https://github.com/mit-acl/faster#user-content-credits:~:text=Issues%20when%20installing%20Gurobi%3A)

akshitj1, 2020-12-03:
For now I have quick fixed with this:
```
find_library( GUROBI_LIBRARY 
              NAMES libgurobi91.so
              PATHS "$ENV{GUROBI_HOME}/lib" 
                    "/Library/gurobi502/mac64/lib"
                    "C:\\libs\\gurobi502\\lib"
              )

find_library( GUROBI_CXX_LIBRARY 
              NAMES gurobi_c++
              PATHS "$ENV{GUROBI_HOME}/lib" 
                    "/Library/gurobi502/mac64/lib"
                    "C:\\libs\\gurobi502\\lib"
              )
```
I will send PR with more robust fix later.

jtorde, 2020-12-03:
Perfect, thanks. I've just updated the FindGurobi.cmake with the one available in official page of gurobi ([link](https://support.gurobi.com/hc/en-us/articles/360039499751-How-do-I-use-CMake-to-build-Gurobi-C-C-projects-)), and I've modified it to make it independent of the version of Gurobi. 

Feel free to create a PR with any better/more robust solutions.