mit-acl/faster #15: Error

xujiahuan, 2020-12-09:
ERROR: cannot launch node of type [global_mapper_ros/global_mapper_node]: Cannot locate node of type [global_mapper_node] in package [global_mapper_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
No processes to monitor

when I run roslaunch global_mapper_ros global_mapper_node.launch...
how can I solve this problem?

jtorde, 2020-12-09:
Hi @xujiahuan!

Some quick questions so that I can help better with this issue:

- Did  you add `source PATH_TO_YOUR_WS/devel/setup.bash`  to your ~/.bashrc, and then run `source ~/.bashrc` in the terminals (or, equivalently, close and open new terminals)?

- What is the output of running `rospack find global_mapper_ros`?

- In your ~/.bashrc, are you doing first `source /opt/ros/melodic/setup.bash` and then `source PATH_TO_YOUR_WS/devel/setup.bash` or the other way around?

jtorde, 2021-01-07:
Closing this for now, feel free to reopen it you still have this issue

closed 2021-01-07 by jtorde