mit-acl/faster #16: Error

xujiahuan, 2020-12-09:
hello developer:
when I run compile the code: catkin build
it shows some problems:

/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h:235:33: error: ‘InternalMetadataWithArena’ in namespace ‘google::protobuf::internal’ does not name a type
   ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h: In member function ‘const google::protobuf::UnknownFieldSet& ignition::msgs::Fog::unknown_fields() const’:
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h:89:12: error: ‘_internal_metadata_’ was not declared in this scope
     return _internal_metadata_.unknown_fields();
            ^~~~~~~~~~~~~~~~~~~
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h: In member function ‘google::protobuf::UnknownFieldSet* ignition::msgs::Fog::mutable_unknown_fields()’:
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h:93:12: error: ‘_internal_metadata_’ was not declared in this scope
     return _internal_metadata_.mutable_unknown_fields();
            ^~~~~~~~~~~~~~~~~~~
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h: In member function ‘google::protobuf::Arena* ignition::msgs::Fog::GetArenaNoVirtual() const’:
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h:131:12: error: ‘_internal_metadata_’ was not declared in this scope
     return _internal_metadata_.arena();
            ^~~~~~~~~~~~~~~~~~~
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h: In member function ‘void* ignition::msgs::Fog::MaybeArenaPtr() const’:
/usr/include/ignition/msgs1/ignition/msgs/fog.pb.h:134:12: error: ‘_internal_metadata_’ was not declared in this scope
     return _internal_metadata_.raw_arena_ptr();
            ^~~~~~~~~~~~~~~~~~~
How to solve this problem? Thanks.

jtorde, 2020-12-09:
Hi @xujiahuan!

This seems a similar error as [this other issue](https://github.com/mit-acl/faster/issues/8).  Specifically, check these versions:

**Ubuntu 18.04 and ROS Melodic**
`protoc --version` should give `libprotoc 3.0.0`
`gazebo --version` should give ` version 9.0.0`

**Ubuntu 16.04 and ROS Kinetic**
`protoc --version` should give `libprotoc 2.6.1`
`gazebo --version` should give ` version 7.0.0`

xujiahuan, 2020-12-10:
Hi jtorde:
Thank you.
I checked these versions:
protoc --version should give libprotoc 3.0.0

xujiahuan, 2020-12-10:
Hi,jtorde:
Thank you.
I checked these versions:
protoc --version is libprotoc 3.14.0
gazebo --version is version 9.16.0
However, these can't solve this problem.

xujiahuan, 2020-12-10:
Hi, developer!
Thank you for your answer!
However,I I had other problems.

No handlers could be found for logger "trollius"ueued] [global_mapper_ros:m... 
_______________________________________________________________________________
Errors     << global_mapper_ros:make /home/hellowu/ws/logs/global_mapper_ros/build.make.000.log
/home/hellowu/ws/devel/.private/global_mapper_ros/lib/libglobal_mapper_ros.so：对‘cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)’未定义的引用  (undefined reference )
/home/hellowu/ws/devel/.private/global_mapper_ros/lib/libglobal_mapper_ros.so：对‘global_mapper::GlobalMapper::PushPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> const> const&)’未定义的引用 (undefined reference )
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/global_mapper_node.dir/build.make:540：/home/hellowu/ws/devel/.private/global_mapper_ros/lib/global_mapper_ros/global_mapper_node] 错误 1
make[1]: *** [CMakeFiles/Makefile2:105：CMakeFiles/global_mapper_node.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
cd /home/hellowu/ws/build/global_mapper_ros; catkin build --get-env global_mapper_ros | catkin env -si  /usr/bin/make --jobserver-auth=6,7; cd -
...............................................................................
Failed     << global_mapper_ros:make               [ Exited with code 2 ]      
Failed    <<< global_mapper_ros                    [ 28.3 seconds ]        

Does this need OpenCV and PCL? If so, what are the versions? Thans a lot.

jtorde, 2020-12-10:
Hi @xujiahuan!

Yes, you need PCL (which is usually installed when installing [`ros-desktop-full`](http://wiki.ros.org/melodic/Installation/Ubuntu#line-34:~:text=Desktop%2DFull%20Install%3A%20(Recommended)%20%3A) I think). Could you try to install it by running
`sudo apt-get install ros-melodic-perception-pcl`? (change `melodic` with the ROS version you are using)

Let me know if this solves the problem

xujiahuan, 2020-12-11:
Hi @jtorde !
I still can't solve this problem. Do you have other ideas? Thank you.

xujiahuan, 2020-12-12:
Hi @jtorde !
Thank you for you answer. 
I still can't solve these problems on my pc.
However, My classmate successfully Compiled it on his computer. His Ubuntu is 16.04/ROS Kinetic, while mine is 18.04/ROS Melodic. So, Maybe it can't be Compiled on Ubuntu 18.04/ROS Melodic? I just guess. Could you provide me with accurate environment configuration? Thank you very much.

jtorde, 2020-12-12:
Hi @xujiahuan,

FASTER has been tested and compiled successfully both in Ubuntu 16.04/ROS Kinetic and Ubuntu 18.04/ROS Melodic. So it should work with the Ubuntu/ROS version you have.

Are you getting the same error as before? If so, could you try to install as well this package (if not already installed): `sudo apt-get install ros-melodic-perception-pcl`

Let me know if this helps!

jtorde, 2020-12-12:
Or more generally, make sure all the packages that come with the Desktop-Full install of ROS are installed:

`sudo apt install ros-melodic-desktop-full`

jtorde, 2021-01-07:
Closing this for now, feel free to reopen it you still have this issue

closed 2021-01-07 by jtorde