mit-acl/faster #17: ERROR

xdwgood, 2020-12-10:
hi developer!
when i try run `roslaunch faster faster.launch`

i got this error:
`ERROR: cannot launch node of type [faster/faster_node]: Cannot locate node of type [faster_node] in package [faster]. Make sure file exists in package path and permission is set to executable (chmod +x)`

I have run: `source ws/devel/setup.bash`

[image]

@jtorde  could you help me?

xdwgood, 2020-12-10:
fixed

closed 2020-12-10 by xdwgood

yanmiao2, 2021-04-05:
@xdwgood Could you share how you fixed the problem?

jtorde, 2021-04-05:
Hi @yanmiao2,

Could you check if Gurobi and its license are installed correctly? I.e., check the output of running `gurobi.sh` on the terminal. If installed correctly, it should launch the `Gurobi Interactive Shell`

yanmiao2, 2021-04-06:
Yes, it's installed correctly. This issue "process has died" seems to happen randomly for me(Approximately 10% chance). But if I close the terminal and redo the command, everything seems fine. I'm not quite sure whether why it behaves this way.