mit-acl/faster #19: Implement PX4 OFFBOARD

pluto-project-it, 2021-01-20:
I am very interested in learning about this project. 
Hello, I see #14 and I was wondering if someone could kindly show me an example of a controller to translate the trajectories processed by the planner into messages compatible with the OFFBOARD mode of PX4.
Even a similar example can be helpful.
Thank you.

jtorde, 2021-03-02:
Hi @pluto-project-it!

I'm not super familiar with the OFFBOARD mode of PX4, but this is what I'd do: 

I'd create a node (following [this example](https://docs.px4.io/master/en/ros/mavros_offboard.html) that subscribes to the goal produced by FASTER (`/SQ01s/goal`), and then every time a goal for FASTER is received, do something like this

```
    pose.pose.position.x = goal.pos.x;
    pose.pose.position.y = goal.pos.y;
    pose.pose.position.z = goal.pos.z;
/////
    local_pos_pub.publish(pose);
```

If the OFFBOARD mode also allows you to send vel, accel and jerk, then you can use the corresponding fields of the goal message. 

Let me know if this solves the issue!

jtorde, 2021-03-16:
Closing this for now, feel free to reopen if you still have this issue

closed 2021-03-16 by jtorde