mit-acl/faster #20: How to run Faster(aerial) on different maps

yanmiao2, 2021-02-23:
Hi developers, I'm new to ROS, and I'm wondering if Faster(aerial robots) also work on other maps? 

If so, where can we find those maps? And how could we import maps to the current project? Appreciate the help, thanks!

jtorde, 2021-02-25:
Hi @yanmiao2!

When following the instructions in the Readme, (specifically when doing `wstool merge ./faster/faster/install/faster.rosinstall` you will clone the repo [acl_gazebo](https://gitlab.com/mit-acl/lab/acl-gazebo), which contains the package `acl_sim`. And then, when running the command `roslaunch acl_sim start_world.launch`, you will start the gazebo world specified in [this line](https://gitlab.com/mit-acl/lab/acl-gazebo/-/blob/master/acl_sim/launch/start_world.launch#L3).

You can also start other different gazebo worlds by simply changing the name of the world in that line. [Here](https://gitlab.com/mit-acl/lab/acl-gazebo/-/tree/master/acl_sim/worlds) and [here](https://github.com/jtorde/planning_worlds_gazebo) there are more `.worlds` available. 

Hope this helps!

jtorde, 2021-03-02:
I'm closing this for now, feel free to reopen if you still have this issue

closed 2021-03-02 by jtorde