mit-acl/faster #21: Unable to navigate drone in z axis

cht649765744, 2021-03-01:
When we set the goal of the drone such that there is change in z-axis coordinate (say from 0,0,0 to 0,0,10). The drone does not move and the faster algorithm keeps outputting “goal reached". Does the FASTER algorithm consider the z axis?

jtorde, 2021-03-02:
Hi @cht649765744!

Yes, FASTER considers the z axis. Could you try to change [z_max](https://github.com/mit-acl/faster/blob/master/faster/param/faster.yaml#L18) to a big value (50 for example), and then send a goal to z=10?

Let me know if this solves the issue

cht649765744, 2021-03-03:
@jtorde Thank you for the reply!

As you suggest, I changed z_max to 20, and then sent a goal to z=10 by using the command "rostopic pub /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: now, frame_id: "map"}, pose: {position: {x: 10.0, y: 0.0, z: 0.0}, orientation: {w: 1.0}}}'
" (Please correct me if this command is wrong.)

But it seems the drone didn't fly up and stayed still, claiming it has already reached the goal. It is what supposed to happen? 

[image]

jtorde, 2021-03-03:
Hi @cht649765744!

Thanks for the detailed explanation. Yes, this shouldn't happen. I'll try to look at it today or tomorrow, and will get back to you.

jtorde, 2021-03-06:
Hi @cht649765744!

Could you try the following? (It's working now on my end after some modifications)

Git pull the last commit of this repo, and then do the wstool merge and wstool update commands as explained in the Readme.

Then, change these parameters:

In `faster.yaml`:
```
force_goal_height: false
goal_height: 1 
z_max: 20 
```

In `global_mapper.yaml`:
```
world_dimensions: [10.0, 10.0, 22.0]
r1: 4
z_max_unknown: 20
```

Then, and after launching everthing as specified in the Readme, you can move the drone to `z=10` by running:
```
rostopic pub /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: now, frame_id: "map"}, pose: {position: {x: 0.0, y: 0.0, z: 10.0}, orientation: {w: 1.0}}}'
```

And later to x=10, y=0, z=10 by running
```
rostopic pub /move_base_simple/goal geomemsgs/PoseStamped '{header: {stamp: now, frame_id: "map"}, pose: {position: {x: 10.0, y: 0.0, z: 10.0}, orientation: {w: 1.0}}}'
```

Note that the camera of the drone is pointing forward. That means that the space above the drone (when moving upwards) will always be unknown space. The "hacked" solution is to modify the r1 parameter (as explained above), so that everything around the drone and inside a sphere of radius r1 is considered known. A better solution would be to modify the orientation of the camera, or add several cameras.  

Hope this helps!

jtorde, 2021-03-16:
Closing this for now, feel free to reopen if you still have this issue

closed 2021-03-16 by jtorde