mit-acl/faster #22: Issue with ROS noetic

Mahathi1992, 2021-03-05:
[build.make.005.log](https://github.com/mit-acl/faster/files/6092561/build.make.005.log)

Attaching the build log file. The create_random_forest.cpp file generates a boost library error. There are a lot of issues with noetic, boost, and Ubuntu 20.04. Is there a way to remove these dependencies?

jtorde, 2021-03-06:
Hi @Mahathi1992! 

I don't have ROS noetic/Ubuntu 20.04 right now, but could you try to `git pull` the last commit of this repo, then do the wstool merge and wstool update commands as explained in the Readme, and then try to compile again?

I changed some things in the `CMakelist.txt` of acl_gazebo

Let me know if this solves the issue!

Mahathi1992, 2021-03-08:
Hi @jtorde, 

I tried your fix and some more steps but Ubuntu 20.04 is not budging. For now I have reverted back to 18.04 and ROS melodic. FASTER should work with this combination right?

jtorde, 2021-03-08:
Hi @Mahathi1992,
 
Yes, Ubuntu 18.04/ROS melodic works