mit-acl/faster #24: Can you help me with a problem happend during building?

zjnyly, 2021-03-23:
I encountered the following problem
`CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find GUROBI (missing: GUROBI_LIBRARY)`

I think I followed all the instructions correctly and the file ~/.bashrc is appended with the following statements:

`source /opt/ros/melodic/setup.bash
export GUROBI_HOME="/home/zjnyly/Desktop/gurobi911/linux64"
export PATH="${PATH}:${GUROBI_HOME}/bin"
export LD_LIBRARY_PATH="${LD_LIBRARY_PATH}:${GUROBI_HOME}/lib"
export GRB_LICENSE_FILE="/home/zjnyly/gurobi.lic"
source /home/zjnyly/Desktop/ws//devel/setup.bash`

I then tried to change the version of gurobi manully in FindGUROBI.cmake file, but I don't know how to correctly write it.
(the version of gurobi on my computer is 9.1.1.

`find_library(GUROBI_LIBRARY
    NAMES gurobi gurobi911
    HINTS ${GUROBI_DIR} $ENV{GUROBI_HOME}
    PATH_SUFFIXES lib)`

I'm new to this field and I don't know much about these things. I'd appreciate it if you could help me. Thanks!

jtorde, 2021-03-31:
Hi @zjnyly,

What is the output of this command (in the terminal): `gurobi.sh` (if correctly installed, the Gurobi Interactive Shell should appear)

I think this problem appears because you are installing Gurobi in `/home/zjnyly/Desktop/gurobi911/`. instead of `/opt/gurobi911/`. Have you tried to install it in `/opt/gurobi911/` instead?

(also, just a minor: it should be
 `source /home/zjnyly/Desktop/ws/devel/setup.bash`, not `source /home/zjnyly/Desktop/ws**//**devel/setup.bash`)

jtorde, 2021-04-06:
Closing this for now, feel free to reopen if you are still facing this issue

closed 2021-04-06 by jtorde