mit-acl/faster #26: No solution found for the safe path, but the obstacle is still far away.

cht649765744, 2021-03-25:
As shown in the following screenshot, after taking off, the drone is given a goal command by  "rostopic pub /move_base_simple/goal geometry_msgs/PoseStamped '{header: {stamp: now, frame_id: "map"}, pose: {position: {x: 10.0, y: 0.0, z: 1.319}, orientation: {w: 1.0}}}'"  However, the drone can not move forward because of  "No solution found for the safe path". But it appears that the obstacle is still far away from the drone. Could you please help to point out the reason why calculating the safe path is infeasible by FASTER? Thanks much!

[image]

[image]

jtorde, 2021-03-31:
Hi @cht649765744,

Are you providing also a depth image to the mapper? From the picture, my guess is that the drone is completely surrounded by unknown space (you can see the unknown space also in Rviz), and therefore there is no safe way to move.

If the unknown space is not important for your specific case, a simple workaround is to publish an unknown point cloud with, let's say, only one point, and far from the robot. It should work then.

cht649765744, 2021-03-31:
Hi @jtorde , thanks for the suggestion on showing unknown space! The following picture is my unknown space. Yes, the depth image has been provided to the mapper, and the mapper has successfully rendered the unknown space and occupancy space. As you can see that there is indeed a known space in front of the drone, this means the drone is not surrounded by the unknown space. With this evidence, could you please provide advice on why the drone can not find a safe way to move? 

[image]

jtorde, 2021-04-02:
Hi @cht649765744, thanks for the detailed explanation. Would it be possible to share the depth image (of rosbag if you prefer) of the camera of the drone at that moment?

Thanks

jtorde, 2021-04-19:
Closing this for now, feel free to reopen if you are still facing this issue

Thanks!

closed 2021-04-19 by jtorde