mit-acl/faster #27: How to programmatically send signals to the Faster GUI?

yanmiao2, 2021-03-31:
Instead of manually clicking on the "Start", "Stop" Emergency Stop" buttons in the Faster GUI every time we start/stop the drone, is there a way to programmatically enable those signals like using a python script?

jtorde, 2021-04-01:
Hi @yanmiao2 ,

You can run `rosservice call /change_mode 'mode: 1'` in the terminal (or from Python/C++ using `system(...)`), and it should have the same effect

jtorde, 2021-04-06:
Closing this for now, feel free to reopen if you are still facing this issue

closed 2021-04-06 by jtorde

yanmiao2, 2021-04-06:
Hi @jtorde, thanks for the reply! I tried the solution you proposed, but am getting this error 
`ERROR: service [/change_mode] responded with an error: service cannot process request: service handler returned None`

Also, just to clarify, is "start" mode 1, "end" mode 2, and "e_stop" mode 3?

yanmiao2, 2021-04-06:
Also a separate question: I'd like to write a function to check the status (E.g. Is destination the obstacle? Did drone collide with walls?...) of the drone at any time.  Is there an existing function/log in faster that I can directly work on?

Sorry for so many questions, since I'm very new to ROS and am trying to figure things out. I deeply the efforts developers invested in this project. Thanks in advance!

jtorde, 2021-04-06:
For the service question, could you try again with the last commit? It's working on my end (I still get the message `service handler returned None` on the terminal, but the drone does take off/land):
`rosservice call /change_mode "mode: 1"` 

The convention is
```
uint8 START  = 1
uint8 END    = 2
uint8 KILL   = 3
```
(see file `MissionMOdeChange.srv`)

For the second question: Right now there are no logs in faster that do those checks directly. One way would be to create a Gazebo plugin to check the collisions, and another one to record a `rosbag` and do a post-analysis using [this](http://wiki.ros.org/rosbag/Code%20API). You can also have a look at [this file](https://github.com/mit-acl/mader/blob/master/mader/scripts/run_many_sims_single_agent.py) (from the MADER repo) to run many simulations and detect the terminal condition. 

Hope this helps!