mit-acl/faster #28: Running the mapper on a real Jackal robot

brian-h-wang, 2021-04-21:
Hi - I'm trying to run the planner on a Jackal robot, following the instructions for running on a real robot, but am having some trouble getting the mapper to work. When I run the mapper (using global_mapper_node.launch), I get the persistent warning

`[ WARN] [1619028102.660679293]: [world_database_master_ros] OnGetTransform failed with "JA01" passed to lookupTransform argument source_frame does not exist.`

and the mapper doesn't publish anything to the occupancy grid topics.

The topics and tf frames published by my Jackal don't use any top-level namespace, which seems like it may be causing the issue (e.g. it just publishes to "odometry/filtered" rather than "JA01/odometry/filtered", or some other namespace that I could use for the "quad" argument in the launch file).  How can I restructure things to make the mapper compatible with my Jackal robot?

Appreciate any help with this, and thanks for your work making this code available!

---

Some more info on my setup: 
- I'm using depth images from a ZED2 stereo camera mounted on the Jackal, and am using a vicon system to get the ground truth pose of the robot.
- The ZED2 node publishes a transform from the "world" frame to the "odom" frame used by the Jackal.
- In the global_mapper_node.launch file, I set the depth_image_topic to the depth image topic published by the ZED2 node, and set the odom_topic to the Jackal's filtered odometry. I'm running the launch file with the quad argument set to JA01. 
- For publishing the robot state: I wrote a script that subscribes to the vicon topic and the Jackal's filtered odometry topic, replaces the pose in the odometry messages with the ground truth pose from the vicon, and republishes the result (as type nav_msgs/Odometry) on the "ground_truth/state" topic. 

Also, here's my tf tree in case this helps: [frames_jackal.pdf](https://github.com/mit-acl/faster/files/6353096/frames_jackal.pdf)

jtorde, 2021-04-22:
Hi @brian-h-wang,

You could probably use it without a namespace (although a namespace is always recommended, specially if you plan to have several robots). Could you try to modify [this line](https://gitlab.com/mit-acl/lab/acl-mapping/-/blob/dev/global-mapper/global_mapper_ros/src/global_mapper_ros/global_mapper_ros.cc#L45) to `name_drone="base_link"`? And then in the launch files use `<arg name="quad" default=""/>`

In case you want to include the namespace in all your topics/tf, check out how it's done [in this package](https://github.com/mit-acl/multi_jackal.git)

Hope this helps!

jtorde, 2021-09-21:
(I'm closing this for now, but feel free to reopen it if you are still having this issue)

closed 2021-09-21 by jtorde