mit-acl/faster #29: setting up environment variable

Ganeshgng, 2021-07-10:
[image]
I followed the guide provided, Using ros-melodic-desktop-full, and gurobi both are working fine.
Aditionally after "wstool merge .........." command I entered "wstool update" but after that no errors showed.
my workspace is in home directory but i can not find "/devel/setup.bash" in there.
so I got error
bash: /home/mrp/ws/devel/setup.bash: No such file or directory

Can you please tell me PATH_TO_YOUR_WS should appear after compiled right? or it will be in any other location?
(I searched devel/setup.bash every folder under /home/mrp/ws.)

Apart from that for the final simulation, NVIDIA GEFORCE 920M is enough?

jtorde, 2021-07-12:
Hi @Ganeshgng,

Did you compile the workspace? The `devel` folder will only appear once you compile the workspace using `catkin build`. 

So the commands would be something like:

`cd /home/mrp/ws/`
`catkin build`

And then you should be able to do `source /home/mrp/ws/devel/setup.bash` in the `bashrc`

And yes,  NVIDIA GEFORCE 920M should be enough

Hope this helps!

Ganeshgng, 2021-07-12:
Thanks for the reply, @jtorde 
yeah I figured out and its working, built success, no errors.

Eventhough I added "source /home/mrp/ws/devel/setup.bash" not working. showed errors.
so I added "source /home/mrp/ws/devel/setup.bash" in the all five terminals before run the roslaunch, and it works.

gazebo.desktop is not responding at the end, taking too much time.
So I can not see the simulation, 
I have Core i7 6th Gen, 8 GB RAM, 920M NVIDIA GEFORCE.
But I am trying with virtual machine gave 6GB RAM for VM, and assigned GPU as well.
So it might be the issue I guess.

Please close this issue.

closed 2021-07-12 by jtorde

Ganeshgng, 2021-07-20:
Hi, @jtorde 
Can you share what is the exact PC requirments(You tested) for Software Simulation? (Processor, RAM, GPU)
I am able to run the FASTER with an aerial robot and FASTER with a ground robot. But its loading too much.
My simulation is way slower than your simulation(https://youtu.be/L13k44-krcc).
Is there any settings I should change to get faster simulation??
May I please know how to change gazebo worlds? Now I am only testing with Forest environment.

Ganeshgng, 2021-08-24:
Dear @jtorde,
I am trying to reproduce the results produced and evaluate your great work. 
I am new to this, I tried several plots on gazebo plotting utility, and exploring the FASTER.
How can I get the logs of FASTER STATS?, Is this option available in your code?
can you suggest a way how you got the statistics results for Faster given in the research paper.
I will be grateful if you can share me this information

[image]