mit-acl/faster #30: snapstack_msgs::Goal QuadGoal has still a problem

kefran88, 2021-07-17:
Hello MIT-ACL team, 

I've been trying to install FASTER and this issue occured : 
_______________________________________________________________________________
Errors     << faster:make /home/fmukwege/workspace/logs/faster/build.make.001.log
/home/fmukwege/workspace/src/faster/faster/src/faster_ros.cpp: In member function ‘void FasterRos::pubCB(const ros::TimerEvent&)’:
/home/fmukwege/workspace/src/faster/faster/src/faster_ros.cpp:272:47: error: no match for ‘operator=’ (operand types are ‘snapstack_msgs::Goal_<std::allocator<void> >::_p_type {aka geometry_msgs::Point_<std::allocator<void> >}’ and ‘geometry_msgs::Vector3 {aka geometry_msgs::Vector3_<std::allocator<void> >}’)
     quadGoal.p = eigen2rosvector(next_goal.pos);
                                               ^
In file included from /opt/ros/melodic/include/geometry_msgs/PointStamped.h:19:0,
                 from /home/fmukwege/workspace/src/faster/faster/include/faster_ros.hpp:11,
                 from /home/fmukwege/workspace/src/faster/faster/src/faster_ros.cpp:9:
/opt/ros/melodic/include/geometry_msgs/Point.h:22:8: note: candidate: geometry_msgs::Point_<std::allocator<void> >& geometry_msgs::Point_<std::allocator<void> >::operator=(const geometry_msgs::Point_<std::allocator<void> >&)
 struct Point_
        ^~~~~~
/opt/ros/melodic/include/geometry_msgs/Point.h:22:8: note:   no known conversion for argument 1 from ‘geometry_msgs::Vector3 {aka geometry_msgs::Vector3_<std::allocator<void> >}’ to ‘const geometry_msgs::Point_<std::allocator<void> >&’
/opt/ros/melodic/include/geometry_msgs/Point.h:22:8: note: candidate: geometry_msgs::Point_<std::allocator<void> >& geometry_msgs::Point_<std::allocator<void> >::operator=(geometry_msgs::Point_<std::allocator<void> >&&)
/opt/ros/melodic/include/geometry_msgs/Point.h:22:8: note:   no known conversion for argument 1 from ‘geometry_msgs::Vector3 {aka geometry_msgs::Vector3_<std::allocator<void> >}’ to ‘geometry_msgs::Point_<std::allocator<void> >&&’
/home/fmukwege/workspace/src/faster/faster/src/faster_ros.cpp:276:14: error: ‘snapstack_msgs::Goal {aka struct snapstack_msgs::Goal_<std::allocator<void> >}’ has no member named ‘dyaw’
     quadGoal.dyaw = next_goal.dyaw;
              ^~~~
/home/fmukwege/workspace/src/faster/faster/src/faster_ros.cpp:277:14: error: ‘snapstack_msgs::Goal {aka struct snapstack_msgs::Goal_<std::allocator<void> >}’ has no member named ‘yaw’
     quadGoal.yaw = next_goal.yaw;
              ^~~
make[2]: *** [CMakeFiles/faster_node.dir/src/faster_ros.cpp.o] Error 1
make[1]: *** [CMakeFiles/faster_node.dir/all] Error 2
make: *** [all] Error 2
-----------------------------

I saw someone had this issue in issue #23 and thus I have cloned the lastest snapstack_msgs commit (1ecc836425f11acfa9c78f18dd22488b7783a852) but nothing changed

What could be the problem ? 

Best regards,

Frank

jtorde, 2021-07-18:
Hi @kefran88,

I have just compiled it without any problem, following the instructions of the readme. Did you run 
`wstool init
wstool merge ./faster/faster/install/faster.rosinstall
wstool update -j8`

before compilation? Specifically, could you check that the `snapstack_msgs` is at the commit specified [here](https://github.com/mit-acl/faster/blob/master/faster/install/faster.rosinstall#L12)?

Let me know if this helps!

kefran88, 2021-07-22:
Hello @jtorde, 

It solved my problem thank you for that, but another just raised. I think it comes from Gurobi, but i had installed it as required in my  /opt/. 
I drop the error message here.  
`CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::setThreads(int)':
solverGurobi.cpp:(.text+0x85b): undefined reference to `GRBModel::set(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::setVerbose(int)':
solverGurobi.cpp:(.text+0x95b): undefined reference to `GRBModel::set(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::SolverGurobi()':
solverGurobi.cpp:(.text+0x209f): undefined reference to `GRBModel::set(GRB_StringAttr, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::setMaxConstraints()':
solverGurobi.cpp:(.text+0x2cb1): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x2e2f): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x2fa5): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x310f): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x3298): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o:solverGurobi.cpp:(.text+0x341c): more undefined references to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)' follow
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::createVars()':
solverGurobi.cpp:(.text+0x3bf8): undefined reference to `GRBModel::addVar(double, double, double, char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::setConstraintsXf()':
solverGurobi.cpp:(.text+0x42e9): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x4443): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x4579): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::setConstraintsX0()':
solverGurobi.cpp:(.text+0x490f): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x4a01): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o:solverGurobi.cpp:(.text+0x4aeb): more undefined references to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)' follow
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o: In function `SolverGurobi::setPolytopesConstraints()':
solverGurobi.cpp:(.text+0x671f): undefined reference to `GRBModel::addVar(double, double, double, char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x6b10): undefined reference to `GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x812f): undefined reference to `GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x81e3): undefined reference to `GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x8290): undefined reference to `GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
solverGurobi.cpp:(.text+0x833d): undefined reference to `GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/fmukwege/workspace/devel/.private/faster/lib/faster/faster_node] Error 1
make[1]: *** [CMakeFiles/faster_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/fmukwege/workspace/build/faster; catkin build --get-env faster | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
`
Best regards, 

Frank

kefran88, 2021-07-22:
I just re-run the part in the readme and it compiled fine. I am closing the issue, thank you for your help !

closed 2021-07-22 by kefran88