mit-acl/faster #33: a problem happend during building

Tanghaonan1998, 2021-09-12:
I encountered the following problem, but i don't know why,please help me

CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::setThreads(int)’中：
solverGurobi.cpp:(.text+0x2270)：对‘GRBModel::set(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::setVerbose(int)’中：
solverGurobi.cpp:(.text+0x2350)：对‘GRBModel::set(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::setMaxConstraints()’中：
solverGurobi.cpp:(.text+0x391e)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x3c73)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x3fc0)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x4315)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x4681)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o:solverGurobi.cpp:(.text+0x49f0): 跟着更多未定义的参考到 GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::SolverGurobi()’中：
solverGurobi.cpp:(.text+0x542f)：对‘GRBModel::set(GRB_StringAttr, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::setConstraintsXf()’中：
solverGurobi.cpp:(.text+0x5e3c)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x5f9f)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x60f2)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::setConstraintsX0()’中：
solverGurobi.cpp:(.text+0x644e)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x6567)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o:solverGurobi.cpp:(.text+0x6678): 跟着更多未定义的参考到 GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::createVars()’中：
solverGurobi.cpp:(.text+0x7a0d)：对‘GRBModel::addVar(double, double, double, char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
CMakeFiles/faster_node.dir/src/solverGurobi.cpp.o：在函数‘SolverGurobi::setPolytopesConstraints()’中：
solverGurobi.cpp:(.text+0x8d8d)：对‘GRBModel::addVar(double, double, double, char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0x93fa)：对‘GRBModel::addConstr(GRBTempConstr const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0xa27c)：对‘GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0xa331)：对‘GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0xa3d0)：对‘GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
solverGurobi.cpp:(.text+0xa467)：对‘GRBModel::addGenConstrIndicator(GRBVar, int, GRBLinExpr const&, char, double, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)’未定义的引用
collect2: error: ld returned 1 exit status
make[2]: *** [/home/thn/project/cproject/competition/ws/devel/.private/faster/lib/faster/faster_node] Error 1
make[1]: *** [CMakeFiles/faster_node.dir/all] Error 2
make: *** [all] Error 2

jtorde, 2021-09-21:
Hi @Tanghaonan1998,

Just to make sure: did you run the commands shown [here](https://github.com/mit-acl/faster#issues-when-installing-gurobi)?
I.e., 
```
cd /opt/gurobi800/linux64/src/build  #Note that the name of the folder gurobi800 changes according to the Gurobi version
sudo make
sudo cp libgurobi_c++.a ../../lib/
```

Let me know if this solves the issue

jtorde, 2021-10-05:
I'm closing this for now, but feel free to reopen it if you are still finding this issue

closed 2021-10-05 by jtorde

johnwaneboy, 2022-04-27:
i have meet the same problem， so how to solve it [image]

yhy130531, 2022-10-29:
> Hi @Tanghaonan1998,
>
> Just to make sure: did you run the commands shown [here](https://github.com/mit-acl/faster#issues-when-installing-gurobi)? I.e.,
>
> ```
> cd /opt/gurobi800/linux64/src/build  #Note [...]
it's wok for me, thanks!