mit-acl/faster #35: Gazebo+PX4-SITL implementation problem

giorgiovaccarino, 2021-10-01:
I'm having trouble implementing faster on a Gazebo+PX4 SITL simulation.
I have my world and a Iris model.
I implemented a controller in python (option 2 of the architecture) but when I subscribe the /goal topic I get:

[ERROR] [1633099445.913813448]: Client [/Extars_Supervisor] wants topic /goal to have datatype/md5sum [snapstack_msgs/Goal/29f7a5b62089bdabd9ea1780f356bc8b], but our version has [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5]. Dropping connection.

I have deleted the workspace and recompiled several times but to no avail.
On the whole disk as a Goal message definition I have these in my workspace:

./home/robot/extars/extars_ws/devel/.private/snapstack_msgs/lib/python2.7/dist-packages/snapstack_msgs/msg/_Goal.py
./home/robot/extars/extars_ws/devel/lib/python2.7/dist-packages/snapstack_msgs/msg/_Goal.py

 Does anyone have an idea?

jtorde, 2021-10-01:
Hi @giorgiovaccarino,

Could you check if you have any other node (apart from FASTER nodes) publishing the `goal` topic? I.e., run `rostopic info /goal` (or `rostopic info /SQ01s/goal` if you are using a namespace). 

It should say that there only FASTER nodes are publishing to that topic, and that its type is `snapstack_msgs/Goal`?

Let me know if this helps!

giorgiovaccarino, 2021-10-04:
There must be some problem with namespaces in my configuration that I can not understand for now.
To continue with the simulation I have renamed the topic /goal in /goalx and I can go on.
Thanks, I close this issue and open another one soon to ask for information about the yaw contained in the Goal message.
[image]

closed 2021-10-04 by giorgiovaccarino

vinesh-anand, 2025-09-18:
I made that work in PX4 that too with ROS noetic 
this algorithm i got as the suggestion in youtube 
i tried its working