mit-acl/faster #37: faster has died with GRBException

giorgiovaccarino, 2021-10-04:
Hello, during a simulated mission if I provide a waypoint beyond a wall I always get this error.
I have Gurobi version 9.12 installed on my system. Can this be a problem?

[image]
[image]
[image]

jtorde, 2022-12-12:
I am closing this, feel free to reopen if you are still facing this issue

closed 2022-12-12 by jtorde

nvhungv2k, 2024-01-13:
@jtorde Pls reopen this issue, I also catch up it
My workspace: Ubuntu 18.04; ROS melodic; FASTER is on origin/master; Gurobi version version 9.1.0, build v9.1.0rc0
I meet this problem as soon as I run:
`$ roslaunch faster faster.launch`
[image]