mit-acl/faster #41: An error occurred while building

MOYFbeta, 2022-07-08:
Hi developer~

i tried to build the project for ground robot and an error occurs.  then i tried to build for drone and got the same error

<pre><font color="#EF2929"><b>Errors</b></font>    <font color="#CC0000"> &lt;&lt;</font> <font color="#06989A">global_mapper_ros</font>:<font color="#3465A4">make</font> <font color="#EF2929"><b>/home/moyf/Workspace/ws/logs/global_mapper_ros/build.make.000.log</b></font>                                                                                                     
/home/moyf/Workspace/ws/devel/.private/global_mapper_ros/lib/libglobal_mapper_ros.so：对‘global_mapper::GlobalMapper::PushPointCloud(boost::shared_ptr&lt;pcl::PointCloud&lt;pcl::PointXYZI&gt; const&gt; const&amp;)’未定义的引用(undefined reference)
collect2: error: ld returned 1 exit status
make[2]: *** [/home/moyf/Workspace/ws/devel/.private/global_mapper_ros/lib/global_mapper_ros/global_mapper_node] Error 1
make[1]: *** [CMakeFiles/global_mapper_node.dir/all] Error 2
make[1]: *** 正在等待未完成的任务(Waiting for unfinished missions)....
make: *** [all] Error 2
<font color="#555753"><b>cd /home/moyf/Workspace/ws/build/global_mapper_ros; catkin build --get-env global_mapper_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -</b></font>
</pre>

I've tried this [#issues-when-installing-gurobi](https://github.com/mit-acl/faster#issues-when-installing-gurobi),it fix other errors but this error remains unfixed

I'm using ROS melodic and gurobi 9.0.3

saurabhprasun20, 2022-09-28:
> Hi developer~
> 
> i tried to build the project for ground robot and an error occurs. then i tried to build for drone and got the same error
> 
> Errors     << global_mapper_ros:make /home/moyf/Workspace/ws/logs/global_mapper_ros/build.make.000.log                                                                                                     
> /home/moyf/Workspace/ws/devel/.private/global_mapper_ros/lib/libglobal_mapper_ros.so：对‘global_mapper::GlobalMapper::PushPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZI> const> const&)’未定义的引用(undefined reference)
> collect2: error: ld returned 1 exit status
> make[2]: *** [/home/moyf/Workspace/ws/devel/.private/global_mapper_ros/lib/global_mapper_ros/global_mapper_node] Error 1
> make[1]: *** [CMakeFiles/global_mapper_node.dir/all] Error 2
> make[1]: *** 正在等待未完成的任务(Waiting for unfinished missions)....
> make: *** [all] Error 2
> cd /home/moyf/Workspace/ws/build/global_mapper_ros; catkin build --get-env global_mapper_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
> I've tried this [#issues-when-installing-gurobi](https://github.com/mit-acl/faster#issues-when-installing-gurobi),it fix other errors but this error remains unfixed
> 
> I'm using ROS melodic and gurobi 9.0.3

If you were able to resolve the issue? I faced the same issue..

nss6planner, 2023-06-25:
Look carefully at the readme given by the author, there are solutions and common errors below, I got the solution after referring

cd /opt/gurobi800/linux64/src/build  #Note that the name of the folder gurobi800 changes according to the Gurobi version
sudo make
sudo cp libgurobi_c++.a ../../lib/