mit-acl/faster #45: Faster in 3D

karolin15, 2023-01-04:
Hi,

Is it possible to use faster as a 3D planning algorithm for a drone, so that it can evade obstacles vertically? I have tried faster in a couple of worlds and it seems like it plans only in a plane towards the goal, and doesn't deviate it's height from that plane? I have used faster with the default values as given in the faster.yaml file, and only changed the maximum z coordinate (z_max) to a height that fits to my worlds. 

Best wishes,
Karolin

jtorde, 2023-01-04:
Hi @karolin15!

FASTER works completely in 3D. It will work in 2D only if you set z_max very similar to z_ground. For the [default values](https://github.com/mit-acl/faster/blob/master/faster/param/faster.yaml#L18) available in faster.yaml, FASTER works in 3D. You can easily test this by introducing obstacles that force the drones to go up and down, or by setting a goal whose height is different than the starting location.

Let me know if this solves the issue!

karolin15, 2023-01-04:
Hi @jtorde, thanks for the fast reply!!

I use worlds that contain multiple lined-up cube-shaped blocks with varying height of 6 to 200 meters - there is also varying space between the buildings where the drone could fly through. I have already set z_max to a very high value (500), so that should not be the issue. The starting and goal height is 5 meters. I change the goal via the terminal, as described in one of the issues #12 . The maximum flown height does not exceed 8 meters in my worlds ( ok yes, then obviously it can plan in 3D - my wording was probably not very precise :) ). This indicates that faster uses a strong preference towards horizontal evasion. Is that correct? I am trying to understand if the behavior that I see is right, or if I have to change any parameters or settings.

( I also changed the parameter force_goal_height to false, as I am changing the goal via terminal. I am not sure if that's correct? )