mit-acl/faster #58: Runtime Error

ltf1001, 2024-07-09:
When I am running the second command：roslaunch acl_sim perfect_tracker_and_sim.launch, the following error occurs:. logging to /home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/roslaunch-dell-2159739.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global symbol: pi
when processing file: /home/oem/faster/src/acl-gazebo/acl_sim/urdf/quadrotor_base.urdf.xacro
included from: /home/oem/faster/src/acl-gazebo/acl_sim/urdf/quadrotor_with_asus.urdf.xacro
included from: /home/oem/faster/src/acl-gazebo/acl_sim/urdf/quadrotor_with_asus.gazebo.xacro
started roslaunch server http://dell:35987/

SUMMARY
========

PARAMETERS
 * /SQ01s/base_link_frame: /base_link
 * /SQ01s/perfect_tracker/x: 0
 * /SQ01s/perfect_tracker/y: 0
 * /SQ01s/perfect_tracker/yaw: 0.0
 * /SQ01s/perfect_tracker/z: 0
 * /SQ01s/robot_description: <?xml version="1....
 * /SQ01s/tf_prefix: 
 * /SQ01s/world_frame: world
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    camera2body (tf/static_transform_publisher)
  /SQ01s/
    perfect_tracker (acl_sim/perfect_tracker.py)
    spawn_robot (gazebo_ros/spawn_model)
    vicon2world (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[SQ01s/spawn_robot-1]: started with pid [2159785]
process[camera2body-2]: started with pid [2159786]
process[SQ01s/vicon2world-3]: started with pid [2159787]
process[SQ01s/perfect_tracker-4]: started with pid [2159788]
  File "/home/oem/faster/src/acl-gazebo/acl_sim/scripts/perfect_tracker.py", line 172
    print "Starting perfect tracker node for: " + ns
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Starting perfect tracker node for: " + ns)?
[SQ01s/perfect_tracker-4] process has died [pid 2159788, exit code 1, cmd /home/oem/faster/src/acl-gazebo/acl_sim/scripts/perfect_tracker.py __name:=perfect_tracker __log:=/home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/SQ01s-perfect_tracker-4.log].
log file: /home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/SQ01s-perfect_tracker-4*.log
[INFO] [1720530213.457985, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1720530213.462070, 64.736000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1720530213.464627, 64.739000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1720530213.726255, 64.849000]: Spawn status: SpawnModel: Successfully spawned entity
[SQ01s/spawn_robot-1] process has finished cleanly
log file: /home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/SQ01s-spawn_robot-1*.log
When I am running the fourth command：roslaunch faster faster_interface.launch, the following error occurs:... logging to /home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/roslaunch-dell-2160001.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dell:37961/

SUMMARY
========

PARAMETERS
 * /SQ01s/faster_commands/is_ground_robot: False
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    rqt_gui (rqt_gui/rqt_gui)
    rviz (rviz/rviz)
  /SQ01s/
    faster_commands (faster/faster_commands.py)

ROS_MASTER_URI=http://localhost:11311

process[SQ01s/faster_commands-1]: started with pid [2160039]
process[rviz-2]: started with pid [2160040]
process[rqt_gui-3]: started with pid [2160041]
  File "/home/oem/faster/src/faster/faster/scripts/faster_commands.py", line 55
    print "Not initialized yet"
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Not initialized yet")?
[SQ01s/faster_commands-1] process has died [pid 2160039, exit code 1, cmd /home/oem/faster/src/faster/faster/scripts/faster_commands.py __name:=faster_commands __log:=/home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/SQ01s-faster_commands-1.log].
log file: /home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/SQ01s-faster_commands-1*.log
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/rqt_gui/rqt_gui", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/main.py", line 61, in main
    return super(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/main.py", line 614, in main
    perspective_manager.import_perspective_from_file(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/perspective_manager.py", line 360, in import_perspective_from_file
    self._convert_values(data, self._import_value)
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/perspective_manager.py", line 422, in _convert_values
    self._convert_values(groups[group], convert_function)
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/perspective_manager.py", line 422, in _convert_values
    self._convert_values(groups[group], convert_function)
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/perspective_manager.py", line 422, in _convert_values
    self._convert_values(groups[group], convert_function)
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/perspective_manager.py", line 419, in _convert_values
    keys[key] = convert_function(keys[key])
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/perspective_manager.py", line 429, in _import_value
    return QByteArray.fromHex(eval(value['repr(QByteArray.hex)']))
  File "<string>", line 1, in <module>
TypeError: arguments did not match any overloaded call:
  QByteArray(): too many arguments
  QByteArray(int, str): argument 1 has unexpected type 'str'
  QByteArray(Union[QByteArray, bytes, bytearray]): argument 1 has unexpected type 'str'
[rqt_gui-3] process has died [pid 2160041, exit code 1, cmd /opt/ros/noetic/lib/rqt_gui/rqt_gui --perspective-file /home/oem/faster/src/faster/faster/rqt_cfgs/demo_interface.perspective __name:=rqt_gui __log:=/home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/rqt_gui-3.log].
log file: /home/oem/.ros/log/9e1b81ca-3df3-11ef-9d53-b3426cc4fe7f/rqt_gui-3*.log

How should I fix the following error? I would greatly appreciate your response.

jtorde, 2024-07-11:
Hi @ltf1001,

FASTER was coded for Python 2.7. The error you are getting 

> SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Not initialized yet")?

is because you are probably using it with Python 3. Try to 

- Either call FASTER with Python 2.7
or
- Add parenthesis to all the prints (see [this ](https://www.datacamp.com/blog/python-2-vs-3-everything-you-need-to-know#:~:text=2%20and%203!-,The%20new%20print()%20function,-The%20print%20statement)for more info)

Feel free to do a pull request if you end up doing the second option

Hope this helps

ltf1001, 2024-07-12:
Thank you for your help, this is a very good method. I have another question, and I really hope to get your help to help me solve it. I saw in your file that you used DecompROS, and you optimized it efficiently in the file. It can achieve the rapid construction of safe corridors in a dynamic three-dimensional environment. However, I am a novice and only understand part of your method, and cannot learn the key methods in your code. How should I learn and what methods should I use to achieve the rapid construction of three-dimensional safe corridors, like you. This work is really perfect, and I think it has a very good inspiration for the rapid planning of mobile robots. I hope to get your help, thank you very much for your help. I am very grateful to you.